toitlang/jaguar #27: Make the application use triggers

krzychb, 2021-12-12:
Resetting the device causes it to forget about the installed application. It is still in flash though, so we could change the code to find it there (maybe keep track of its position in NVS?) and automatically start it. 

Ideally, the application would use triggers from .yaml file - https://docs.toit.io/platform/apps/appspec/#the-specification-file

Inspired by https://esp32.com/viewtopic.php?f=2&t=24941#p88621

kasperl, 2021-12-12:
It would make sense to implement a way to restart the last installed app to make testing of 'deep sleep' and GPIO-based wakeup behavior possible.

Jaguar is primarily thought of as a development (not deployment) framework, so more complex deployment support with .yaml files, etc. might be out of the initial scope.

kasperl, 2021-12-23:
I think we can get pretty far by storing the last program (offset, size) in `device.FlashStore` and re-run it on boot. Maybe something for you, @lask?

We should make sure to clear the program before we kill it and (potentially) overwrite it in flash. And we should write the last program information just before starting the program when everything is ready to go.

PR #89 (2021-12-29): Re-run last installed program on reboot.
Fixes #27.
commit 1814e22e (2021-12-29): Re-run last installed program on reboot.

Fixes #27.
closed 2021-12-29 by kasperl