toitlang/jaguar #28: Provide a way to run the applications from individual containers

krzychb, 2021-12-12:
Jaguar works by loading the application into a separate container, but it only runs one at a time. We start installing the program here https://github.com/toitlang/jaguar/blob ... r.toit#L49 (this is what kills the old version of the program) and we start running it here https://github.com/toitlang/jaguar/blob ... r.toit#L60.

This issue is suggesting to implement a new feature to load the applications into individual containers identified by the `name: <App name>` from https://docs.toit.io/platform/apps/appspec/#the-specification-file

Inspired by https://esp32.com/viewtopic.php?f=2&t=24941#p88621

kasperl, 2021-12-12:
Being able to develop apps that run across multiple containers using Jaguar seem useful.

kasperl, 2022-08-30:
@krzychb Finally made some progress with this. Once #228 lands, you can install multiple containers and have them run side by side using:

``` sh
$ jag container install hello examples/hello.toit
$ jag container install http examples/http/http.toit
$ jag container list
718b6ac6-860c-58af-8ef7-f694e7284dc6: jaguar
0b777bc9-d58c-56b2-a030-1738f48888b1: hello
6173c8b0-be07-5b2e-852a-1e9cc5847a3a: http
```

closed 2022-08-30 by kasperl

krzychb, 2022-08-30:
Thank you @kasperl!