toitlang/jaguar #30: 'jag simulate' fails on windows

jstroem, 2021-12-13:
'jag simulate' fails with error:

```
> jag simulate
Error: exit status 0xc0000135

```

kasperl, 2021-12-14:
Fixed in https://github.com/toitlang/toit/commit/53399f84edb75c55695beace4909dae17e5e956e which is part of SDK v0.0.5.

jstroem, 2021-12-14:
Still a problem:
```
>jag decode --system WyNVBVVYU1UAU1UAWyRVI1UQAAAAAAAAAAAAAAAAAAAAAFsjVQRVRVNVF1BSSU1JVElWRV9MT09LVVBfRkFJTEVEU1UcRmFpbGVkIHRvIGZpbmQgcHJpbWl0aXZlIDI6MFsjVQJVU1sjVRBbI1UDVUZVAEkivlsjVQNVRlUBSQE/WyNVA1VGVQJJFexbI1UDVUZVA0kWGlsjVQNVRlUESSpHWyNVA1VGVQVJfixbI1UDVUZVBkkCoFsjVQNVRlUHSQ2jWyNVA1VGVQhJGMdbI1UDVUZVCUlEJFsjVQNVRlUKSQ6ZWyNVA1VGVQtJDq5bI1UDVUZVDEkOXlsjVQNVRlUNSUQ3WyNVA1VGVQ5JRExbI1UDVUZVD0kNyA==
PRIMITIVE_LOOKUP_FAILED error.
Failed to find primitive 2:0
  0: tcp_init_                 <sdk>/net/modules/tcp.toit:164:1
  1: tcp_resource_group_       <sdk>/net/modules/tcp.toit:160:25
  2: run_global_initializer_.<block> <sdk>/core/objects.toit:205:26
  3: run_global_initializer_   <sdk>/core/objects.toit:194:1
  4: TcpServerSocket.listen    <sdk>/net/modules/tcp.toit:92:24
  5: SystemInterface_.tcp_listen <sdk>/net/impl.toit:48:12
  6: main                      /home/runner/work/jaguar/jaguar/src/jaguar.toit:50:21
  7: __entry__.<lambda>        <sdk>/core/entry.toit:48:20
```

but was known, we haven't implemented network on windows.

kasperl, 2022-11-27:
I believe this has been fixed.

closed 2022-11-27 by kasperl