toitlang/jaguar #47: Image writer out of bounds

lask, 2021-12-15:
Got the following exception when using Jag:
```
$ jag decode --system WyNVBVVYU1UAU1UAWyRVI1UQAAAAAAAAAAAAAAAAAAAAAFsjVQRVRVNVCUVYQ0VQVElPTlNVDU9VVF9PRl9CT1VORFNbI1UCVVNbI1UZWyNVA1VGVQBJCNVbI1UDVUZVAUl7AlsjVQNVRlUCSXybWyNVA1VGVQNJA9BbI1UDVUZVBEkB1VsjVQNVRlUFSUV3WyNVA1VGVQZJDs5bI1UDVUZVB0kO41sjVQNVRlUISQ7AWyNVA1VGVQlJRcBbI1UDVUZVCklEglsjVQNVRlULSQ7OWyNVA1VGVQxJDuNbI1UDVUZVDUkOk1sjVQNVRlUOSUStWyNVA1VGVQ9JRPtbI1UDVUZVEEkY+lsjVQNVRlURSUm9WyNVA1VGVRJJDs5bI1UDVUZVE0kO41sjVQNVRlUUSQ6TWyNVA1VGVRVJSdBbI1UDVUZVFklJ5VsjVQNVRlUXSRifWyNVA1VGVRhJGRM=
EXCEPTION error. 
OUT_OF_BOUNDS
  0: image_writer_commit_      /home/runner/work/jaguar/jaguar/src/programs.toit:130:3
  1: FlashImageWriter_.commit  /home/runner/work/jaguar/jaguar/src/programs.toit:106:5
  2: ProgramManager.commit     /home/runner/work/jaguar/jaguar/src/programs.toit:41:13
  3: install_program           /home/runner/work/jaguar/jaguar/src/jaguar.toit:73:22
  4: main.<lambda>             /home/runner/work/jaguar/jaguar/src/jaguar.toit:58:7
  5: Server.run_connection_.<block> <pkg:pkg-http>/server.toit:81:17
  6: catch.<block>             <sdk>/core/exceptions.toit:114:10
  7: catch                     <sdk>/core/exceptions.toit:112:1
  8: catch                     <sdk>/core/exceptions.toit:87:10
  9: Server.run_connection_    <pkg:pkg-http>/server.toit:80:7
 10: Server.listen.<lambda>.<block>.<block> <pkg:pkg-http>/server.toit:65:13
 11: catch.<block>             <sdk>/core/exceptions.toit:114:10
 12: catch                     <sdk>/core/exceptions.toit:112:1
 13: catch                     <sdk>/core/exceptions.toit:63:10
 14: Server.listen.<lambda>.<block> <pkg:pkg-http>/server.toit:64:16
 15: Server.listen.<lambda>    <pkg:pkg-http>/server.toit:52:24
```

kasperl, 2021-12-17:
Since jaguar v0.1.3, the image writer code now produces more recognizable exceptions. Let me know if you see this again and please do upload a new stack trace in that case.

kasperl, 2022-02-07:
Fixed in SDK v1.6.0.

closed 2022-02-07 by kasperl