toitlang/jaguar #87: tried jag and get stuck

mess-maker, 2021-12-25:
hi
tried jaguar following tutorial on an esp32 I have lying around DFRobot [DFR0478](https://www.dfrobot.com/product-1590.html)

```jag setup```
setup seems ok

```
jag flash                                                                                    
Enter WiFi network (SSID): My_SSID
Enter WiFi password for 'Livebox-6070': ******
Flashing device over serial on port '/dev/ttyUSB0' ...
esptool.py v3.0
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32-D0WD (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: f0:08:d1:65:c8:d0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 16MB
Flash params set to 0x0240
Compressed 15408 bytes to 10785...
Wrote 15408 bytes (10785 compressed) at 0x00001000 in 0.1 seconds (effective 848.8 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 119...
Wrote 3072 bytes (119 compressed) at 0x00008000 in 0.0 seconds (effective 4731.9 kbit/s)...
Hash of data verified.
Compressed 1091696 bytes to 719903...
Wrote 1091696 bytes (719903 compressed) at 0x00010000 in 11.1 seconds (effective 788.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```
here looks good too and the device appear in my router device list

```
jag monitor                                                                                   
Starting serial monitor of port '/dev/ttyUSB0' ...
����������������������������������ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:188
ho 0 tail 12 room 4
load:0x40078000,len:12180
load:0x40080400,len:2944
entry 0x400805c8
clearing RTC memory: RTC memory is in inconsistent state
E (463) esp_ota_ops: not found otadata
[flash reg] address 0x3f430000, size 0x00200000
[wifi] DEBUG: connecting
[wifi] DEBUG: connected
[wifi] INFO: got ip ip=192.168.1.20
INFO: Running Jaguar device 'sad-lorry' (id: '3ea2d2ee-54fd-4f05-bf25-5840d9381f89') on 'http://192.168.1.20:9000'
```
some weird thing here I think

and finally got stuck there :/ (monitor was still active)
```
jag scan                                                                                      
Scanning ...
Error: didn't find any Jaguar devices
```

Is my device dead ?
Or maybe I do something wrong ?

mess-maker, 2021-12-25:
hi,

checked the device by sending a blink arduino sketch via platformio
device work ok here

mess-maker, 2021-12-25:
some more info

OpenSUSE Tumbleweed vm on win10 host using VMware player (network configured by default in NAT on vmware player)

jag scan work ok when used from win10 host
opening 1990/UDP port on linux guest firewall don't solve the issue
disabling totally the firewall on linux guest don't solve the issue either

```
❯ sudo systemctl status firewalld
○ firewalld.service - firewalld - dynamic firewall daemon
     Loaded: loaded (/usr/lib/systemd/system/firewalld.service; disabled; vendor preset: disabled)
     Active: inactive (dead)
       Docs: man:firewalld(1)

❯ jag scan                                                                                                                                              
Scanning ...
Error: didn't find any Jaguar devices
```

mess-maker, 2021-12-25:
tested also on a fedora 35 VM
1990/UDP is open by default but jag scan fail
disabling firewall give same result

mess-maker, 2021-12-26:
done some more googling
problem seems related to the ''NATed'' setup of the guest/host network configuration 
https://superuser.com/questions/822422/can-a-vmware-player-virtual-machine-receive-udp-packages
the UDP broadcast don't reach the VM.
I'm stuck to stay on windows as VMWare player (non pro) don't allow to modify the NAT routing or have to switch to a bridged network setup

maybe you can close this issue or notify in the ''how's to'' that IT may have issue when using a VM

kasperl, 2021-12-27:
Filed https://github.com/toitlang/jaguar/issues/88 to make it easier to connect to a specific device instead of waiting for broadcasted UDP packages to reach the host. Closing this issue, but thank @mess-maker for filing it!

closed 2021-12-27 by kasperl