toitlang/jaguar #136: Threading issue with exit code.

floitsch, 2022-02-17:
Sometimes "process terminated - exit code: 0" is before the output of the program:
```
[3.toit] 2022-02-17T21:21:07.057094Z: <process initiated>
[3.toit] There are 8 elements in the list
[3.toit] Here they are (sorted):
[3.toit] 0. Baboon
[3.toit] 1. Cat
[3.toit] 2. Fish
[3.toit] 3. Helicopter
[3.toit] 4. Horse
[3.toit] 5. Mouse
[3.toit] 6. Penguin
[3.toit] 2022-02-17T21:21:07.557674Z: <process terminated - exit code: 0>
[3.toit] 7. Radish
[3.toit] list is Baboon, Cat, Fish, Helicopter, Horse, Mouse, Penguin, Radish
```

kasperl, 2022-06-30:
I have never seen this. I wonder what is going on?

floitsch, 2022-06-30:
This looks like a simulator issue. It was reported with `toit run --no-device ...`
Closing.

closed 2022-06-30 by floitsch