toitlang/jaguar #145: setup wants to install release 1.0.8 which is not there yet

lutzbickhardt, 2022-03-04:


lutzbickhardt, 2022-03-04:
lutzbickhardt_27@Lutzs-iMac jaguar % $JAG_EXE version
Version:	 v1.0.8
SDK version:	 v1.6.8
Build date:	 2022-03-02T05:01:06Z
lutzbickhardt_27@Lutzs-iMac jaguar % $JAG_EXE setup  
Downloading Toit SDK from https://github.com/toitlang/toit/releases/download/v1.6.8/toit-macos.tar.gz ...
18.34 MiB / 18.34 MiB [-------------------------------------------------------------------------------------------------------------------] 100.00% 1.78 MiB p/s
Successfully installed Toit SDK into /Users/lutzbickhardt_27/.cache/jaguar/sdk
Downloading ESP32 image from https://github.com/toitlang/jaguar/releases/download/v1.0.8/image.tar.gz ...
Error: failed downloading the Toit SDK artifact: 404 Not Found

floitsch, 2022-03-06:
The 1.0.8 requirement was added to master, but we haven't yet released Jaguar with that requirement.
How did you install Jaguar?

lutzbickhardt, 2022-03-06:
https://github.com/toitlang/jaguar?

Cordialement

--------------------------
Lutz BICKHARDT
***@***.***

Graumannsweg 42, D-22087 Hamburg, Germany

SKYPE: bickhardt

+49 162 9784296
+49 172 3846086
+49 40 2275812

closed 2022-03-07 by floitsch

lutzbickhardt, 2022-03-07:
actually I downloaded noth using go but git:

- git clone --recursive https://github.com/toitlang/jaguar.git

and then

export IDF_PATH=/Users/lutzbickhardt_27/jaguar/third_party/toit/third_party/esp-idf
export TOIT_PATH=/Users/lutzbickhardt_27/jaguar/third_party/toit
export JAG_PATH=/Users/lutzbickhardt_27/jaguar
cd $TOIT_PATH

$IDF_PATH/install.sh
source $IDF_PATH/export.sh
make all

$TOIT_PATH/build/host/sdk/bin/toit.pkg install --project-root=$JAG_PATH
cd $JAG_PATH

make

export JAG_EXE=/Users/lutzbickhardt_27/jaguar/build/jag

and then:

 $JAG_EXE setup?

Since the source at github does not seem to have been changed, I guess the problem for me is not solved (I did >make clean<, >make< to rebuild the executable though, but same result)

Cordialement

--------------------------
Lutz BICKHARDT
***@***.***

Graumannsweg 42, D-22087 Hamburg, Germany

SKYPE: bickhardt

+49 162 9784296
+49 172 3846086
+49 40 2275812

Closed #145<https://github.com/toitlang/jaguar/issues/145>.

-

floitsch, 2022-03-07:
If you build yourself, then you don't need to run `setup`.

The `setup` command downloads the image and tools, that you just built in the `TOIT_PATH` folder.
You just need to set `JAG_TOIT_PATH` and `JAG_ENTRY_POINT` (as described in the README) and things should work.

lutzbickhardt, 2022-03-07:
?they do indeed, thanks a lot ;-))

Cordialement

--------------------------
Lutz BICKHARDT
***@***.***

Graumannsweg 42, D-22087 Hamburg, Germany

SKYPE: bickhardt

+49 162 9784296
+49 172 3846086
+49 40 2275812

lutzbickhardt, 2022-03-09:
actually the problem was that the VERSION file was not built/copied;

cp /Users/lutzbickhardt_27/.cache/jaguar/sdk/VERSION /Users/lutzbickhardt_27/jaguar/third_party/toit/build/host/sdk/VERSION?

solved it !

Cordialement

--------------------------
Lutz BICKHARDT
***@***.***

Graumannsweg 42, D-22087 Hamburg, Germany

SKYPE: bickhardt

+49 162 9784296
+49 172 3846086
+49 40 2275812

________________________________
De : Bickhardt, Lutz
Envoyé : lundi 7 mars 2022 14:42
À : toitlang/jaguar
Objet : RE: [toitlang/jaguar] setup wants to install release 1.0.8 which is not there yet (Issue #145)

?they do indeed, thanks a lot ;-))

Cordialement

--------------------------
Lutz BICKHARDT
***@***.***

Graumannsweg 42, D-22087 Hamburg, Germany

SKYPE: bickhardt

+49 162 9784296
+49 172 3846086
+49 40 2275812

kasperl, 2022-03-09:
Thanks for reporting this, @lutzbickhardt! I think I've fixed this issue (missing VERSION file) in https://github.com/toitlang/toit/pull/550.