toitlang/jaguar #161: Error: didn't find any Jaguar devices

kaxori, 2022-05-16:
From tmit to time Jaguar application stops running:

  failed to ping 'device'.
  Scanning ...
  Error: didn't find any Jaguar devices

[wifi] DEBUG: connecting
[wifi] DEBUG: connected
[wifi] DEBUG: closing
[jaguar] WARN: running Jaguar failed due to 'DEADLINE_EXCEEDED' (2/3)

??? any glue?

floitsch, 2022-05-16:
Are there any exceptions that could help debug this issue?

kaxori, 2022-05-17:
this is the monitor output:

o
o    ESP32 TOIT Prototype, Motion sensor and HASS-interface
o
o    dev-ID: 1f84d167-d56b-5c38-85f4-610f775a97a8
o

[wifi] DEBUG: connecting
[wifi] DEBUG: connected
[wifi] DEBUG: closing
[wifi] DEBUG: connecting
[jaguar] WARN: running Jaguar failed due to 'DEADLINE_EXCEEDED' (1/3)
[wifi] DEBUG: connected
[wifi] DEBUG: closing
[wifi] DEBUG: connecting
----
Received a Toit system message. Executing the command below will
make it human readable:
----
build/host/sdk/bin/toit.run tools/system_message.toit "$SNAPSHOT" -b WyNVBVVYU1UOdjIuMC4wLWFscGhhLjFTVQBbJFUjVRAAAAAAAAAAAAAAAAAAAAAAWyNVBFVFU1UJRVhDRVBUSU9OU1URREVBRExJTkVfRVhDRUVERURbI1UCVVNbI1UjWyNVA1VGVQBJWCVbI1UDVUZVAUliSlsjVQNVRlUCSWFlWyNVA1VGVQNJWaNbI1UDVUZVBElaBVsjVQNVRlUFSWF4WyNVA1VGVQZsAACJb1sjVQNVRlUHSTqrWyNVA1VGVQhJRLhbI1UDVUZVCUlE3VsjVQNVRlUKSRj7WyNVA1VGVQtJGNhbI1UDVUZVDEk61lsjVQNVRlUNSTr5WyNVA1VGVQ5JYrlbI1UDVUZVD0lZo1sjVQNVRlUQSVoFWyNVA1VGVRFJYtBbI1UDVUZVEkk6Q1sjVQNVRlUTSTnFWyNVA1VGVRRJOTpbI1UDVUZVFUk5tVsjVQNVRlUWSR1oWyNVA1VGVRdJJaJbI1UDVUZVGEklylsjVQNVRlUZSVz9WyNVA1VGVRpJXSVbI1UDVUZVG0ldXFsjVQNVRlUcSQoUWyNVA1VGVR1JCztbI1UDVUZVHkkLUFsjVQNVRlUfSQsrWyNVA1VGVSBJRX5bI1UDVUZVIUlFkVsjVQNVRlUiSQoo
[wifi] DEBUG: connected
[wifi] DEBUG: closing
[jaguar] WARN: running Jaguar failed due to 'DEADLINE_EXCEEDED' (2/3)
[wifi] DEBUG: connecting
[wifi] DEBUG: connected

kaxori, 2022-05-17:
I found out, that after reflashing the controller runs.
But if I start the "jag monitor" afterwards the controller hangs up ans shows the 'DEADLINE_EXCEEDED' message.

After a reset with running/already opened montor the controller runs properly and shows the debug message in monitor window.

BUT if I close the monitor an reopen it again, the controller behaves faulty again.

closed 2022-05-17 by kaxori