toitlang/jaguar #192: hangup with reconnect problem

kaxori, 2022-07-02:
after successfull start and longer run, the toit firmware reports wifi reconnection errors:

**Restart Message:**
[image]

**Decoded Error:**
EXCEPTION error.
DEADLINE_EXCEEDED
  0: __Monitor__.await         <sdk>/core/monitor_impl.toit:28:20
  1: ResourceState_.wait_for_state_ <sdk>/core/events.toit:55:5
  2: ResourceState_.wait.<block> <sdk>/core/events.toit:22:12
  3: __Monitor__.locked_.<block> <sdk>/core/monitor_impl.toit:123:12
  4: __Monitor__.locked_       <sdk>/core/monitor_impl.toit:95:3
  5: ResourceState_.wait       <sdk>/core/events.toit:21:3
  6: WifiModule.wait_for_ip_address system/extensions/esp32/wifi.toit:208:25
  7: WifiServiceDefinition.turn_on.<block>.<block> system/extensions/esp32/wifi.toit:67:47
  8: Task_.with_deadline_.<block> <sdk>/core/task.toit:73:21
  9: Task_.with_deadline_      <sdk>/core/task.toit:67:3
 10: with_timeout              <sdk>/core/utils.toit:179:15
 11: with_timeout              <sdk>/core/utils.toit:162:12
 12: WifiServiceDefinition.turn_on.<block> system/extensions/esp32/wifi.toit:67:7
 13: WifiServiceDefinition.turn_on system/extensions/esp32/wifi.toit:62:3
 14: WifiState.up.<block>.<block> system/extensions/esp32/wifi.toit:102:31
 15: WifiState.up.<block>      system/extensions/esp32/wifi.toit:98:3
 16: __Monitor__.locked_.<block> <sdk>/core/monitor_impl.toit:123:12
 17: __Monitor__.locked_       <sdk>/core/monitor_impl.toit:95:3
 18: WifiState.up              system/extensions/esp32/wifi.toit:98:3
 19: WifiServiceDefinition.connect system/extensions/esp32/wifi.toit:53:23
 20: WifiServiceDefinition.connect system/extensions/esp32/wifi.toit:44:12
 21: NetworkServiceDefinitionBase.handle system/extensions/shared/network_base.toit:26:14
 22: WifiServiceDefinition.handle system/extensions/esp32/wifi.toit:41:12
 23: ServiceManager_.<lambda>  <sdk>/system/services.toit:332:15
 24: RpcRequest_.process.<block> <sdk>/rpc/broker.toit:98:26
 25: RpcRequest_.process       <sdk>/rpc/broker.toit:95:3
 26: RpcRequestQueue_.ensure_processing_task_.<lambda>.<block>.<block> <sdk>/rpc/broker.toit:214:20
 27: RpcRequestQueue_.ensure_processing_task_.<lambda>.<block> <sdk>/rpc/broker.toit:209:9
 28: RpcRequestQueue_.ensure_processing_task_.<lambda> <sdk>/rpc/broker.toit:204:81

kasperl, 2022-07-02:
So it times out trying to get an IP address using DHCP. 

Does rebooting the device help at all? Does it continue to fail for longer periods of time? 

There could be something wrong with the handling of timeouts in that part of the code (wait_for_ip_address), so we should definitely take a look and see if we can reproduce the problem.

kaxori, 2022-07-02:
Wifi connection seems to be lost and for this reason the mqtt send fails ...
auto reconnect trials are not successful
heap seems to get overloaded

[image]

kaxori, 2022-07-02:
after resetting it starts again:
[image]

erikcorry, 2022-07-04:
It's a bit of a guess, but I think there may be two problems here.

Firstly the Wifi disconnects, perhaps because memory is exhausted.  You can limit the
amount of memory your program uses with the experimental call

set_max_heap_size_  64 * 1024

eg to limit your program to 64k.  You may need to experiment to get a size that be sufficient,
but will also collect garbage more agressively, so that the wifi driver does not see memory
exhaustion issues.

The second issue is the failure to reconnect - we will look into that.

kasperl, 2022-08-03:
Hi @kaxori! Did you make any progress on this? We just released new versions of Toit and Jaguar (v2.0.0-alpha.14 and v1.4.1 respectively) and I can tell from your traces that you've seen the issues on v2.0.0-alpha.10.

kasperl, 2022-08-03:
You might be able to benefit from the `print_objects` helper to figure out if your memory usage keeps growing.

```
main:
  iteration := 0
  while true:
    print_objects "iteration=$(iteration++)"
    sleep --ms=1_000
```

kasperl, 2022-09-10:
Hi again @kaxori! Are there any steps for us to reproduce the issue? We just sent out Jaguar v1.5.3 with some GC improvements.

kasperl, 2023-03-11:
We can't reproduce this :(

closed 2023-03-11 by kasperl