toitlang/jaguar #205: Default timeout for 'jag scan' with IP address is too low

kasperl, 2022-08-03:
It seems pretty common that we cannot complete the HTTP /identify request in 600ms. Perhaps we need a higher default timeout when we're scanning specific addresses because it involves a HTTP roundtrip that isn't necessary when just listening for broadcasted UDP packets?

floitsch, 2022-08-03:
600ms sounds like a lot of time. I wonder if something else is going wrong.

addshore, 2025-04-24:
So I often (tens of hundreds of times a day) run into this sort of issue when specifying a device IP..

```
jag container install demo ./main.toit --device 192.168.68.79 -D lb-dev=1
Scanning for device with address: '192.168.68.79'
Error: Get "http://192.168.68.79:9000/identify": context deadline exceeded
```

I know the device is there, and I can even load the identify in a browser for example, but sometimes indeed the request might be slightly sluggish (because http things)

SO I would be very pro a longer timeout if that would eliminate the issue.
My other suggestion would be a `--retry` or such, that would keep trying to identify...

PR #612 (2025-04-28): Increase timeout for identify call.
Fixes #205.
commit 8a6ef0ba (2025-04-28): Increase timeout for identify call. (#612)

Fixes #205.
closed 2025-04-28 by floitsch