toitlang/jaguar #290: ESP32-C3 not support

nne998, 2022-10-09:
% jag flash           
Flashing device over serial on port '/dev/cu.usbserial-2140' ...
esptool.py v3.0
Serial port /dev/cu.usbserial-2140
Connecting....
WARNING: This chip doesn't appear to be a ESP32 (chip magic value 0x1b31506f). Probably it is unsupported by this version of esptool.
Chip is ESP32-D0WDQ6 (revision 0)
Features: WiFi, BT, Dual Core, Coding Scheme None
WARNING: Detected crystal freq 0.00MHz is quite different to normalized freq 26MHz. Unsupported crystal in use?
Crystal is 26MHz
MAC: 00:00:00:00:00:00
Uploading stub...
Running stub...

A fatal error occurred: Invalid head of packet (0x47)
Error: exit status 2

floitsch, 2022-10-09:
Note that C3 support is still alpha. We only committed support for compilation last week. We ordered some C3 boards, but until then it's pretty much untested.

We will add more ESP32-C3 support in the future.

That said, here are some instructions to get the C3 working already now. (Admittedly untested, since I don't have a C3 board yet).

The Jaguar commands are currently hard-coded: https://github.com/toitlang/jaguar/blob/main/cmd/jag/commands/flash.go#L230
And other places where the `esp32` firmware and not `esp32-c3` firmware is referenced.
You could either change the source files or create the firmware with the Jaguar-service yourself and then flash using the esptool.

I modified my Jaguar to print out the commands that it is doing before calling the flash command:
- `assets` is the `sdk/tools/assets` executable.
- `firmware` is the `sdk/tools/firmware` executable.
- `flash` is the esp-idf flash command.
- `/tmp/854298073.%sdataota` is a file that only contains `0xff` bytes.
- `/tmp/3703377921.%sdatanvs`, too, is a file that only contains `0xff` bytes.
The last two files are just there to clear the flash. You shouldn't need them if you erase the flash first.  (And yes: the naming got screwed up. I will write a patch now).

```
assets [-e /tmp/1340211712.assets create]
assets [-e /tmp/1340211712.assets add --format=tison config /tmp/4141227791.json.assets]
firmware [-e /home/flo/work/opentoit/build/esp32/firmware.envelope container install --assets /tmp/1340211712.assets -o /tmp/4191997074.envelope jaguar /home/flo/work/jaguar/build/assets/jaguar.snapshot]
firmware [-e /tmp/4191997074.envelope property set uuid 5f15c969-4fc4-4000-9bbf-9764e8e82c87]
firmware [-e /tmp/4191997074.envelope property set wifi { "wifi.ssid": "wifi 16.3", "wifi.password": "internet1234" }]
firmware [-e /tmp/4191997074.envelope extract --system.snapshot -o /tmp/system.snapshot.2377688526]
firmware [-e /tmp/4191997074.envelope extract --firmware.bin -o /tmp/firmware.bin.2309810457]
firmware [-e /tmp/4191997074.envelope extract --bootloader.bin -o /tmp/bootloader.bin.3342109601]
firmware [-e /tmp/4191997074.envelope extract --partitions.bin -o /tmp/partitions.bin.3870265247]
firmware [-e /tmp/4191997074.envelope extract --partitions.csv -o /tmp/partitions.csv.993326306]
bootloaderBin.Name() /tmp/bootloader.bin.3342109601
Flashing device over serial on port '/dev/ttyUSB0' ...
flash [--chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 /tmp/bootloader.bin.3342109601 0x8000 /tmp/partitions.bin.3870265247 0x10000 /tmp/firmware.bin.2309810457 0xd000 /tmp/854298073.%sdataota 0x350000 /tmp/3703377921.%sdatanvs]
```

If you want to compile against the ESP32-c3 you will need to.

## Create the ESP32-C3 firmware:
Check out the Toit repository (https://github.com/toitlang/toit.git) and compile it for the ESP32-C3:
```
make ESP32_CHIP=esp32c3 esp32
```
This gives you a build/esp32-c3/firmware.envelope.
(Sidenote: during compilation there are currently still some warnings. That's because we haven't really played with the C3 yet).

## Add the Jaguar service to the envelope
I would check out the Jaguar repository and then:
- either change all references to the esp32 with esp32-c3, or
- run the commands from above, just with the esp32-c3 firmware.

Either way, you almost certainly want to set the `JAG_TOIT_REPO_PATH` to the Toit repository you checked out:
```
export JAG_TOIT_REPO_PATH=$PWD/third_party/toit
```
See https://github.com/toitlang/jaguar#building-it-yourself

After that you should be able to use the CLI version of Jaguar to talk to your device.

JWood48, 2023-01-19:
What is the status on esp32-c3 support? are the steps above still necessary?

floitsch, 2023-01-19:
Unfortunately we haven't made a lot of progress on the ESP32 variants. It's still high on our list, but currently the steps are probably necessary.

kasperl, 2023-03-11:
We ship pre-packaged ESP32-C3 firmware now, so you should be able to use `jag flash --chip esp32c3` to flash it. Please take another look, @JWood48!

closed 2023-03-11 by kasperl

JWood48, 2023-03-22:
Did not have time before now, but is is flashing and booting! Thanks!