toitlang/jaguar #306: jag flash not working on windows

JWood48, 2022-11-24:
After installing jaguar 1.7.8 on windows 11 I get the following error when flashing devices;

```
$ jag flash -p COM4
Flashing device over serial on port 'COM4' ...
EXCEPTION error. 
cannot open port 'COM4'
  0: flash                     D:\a\toit\toit\tools\firmware.toit:497:5
  1: flash_cmd.<lambda>        D:\a\toit\toit\tools\firmware.toit:486:16
  2: Command.run               <pkg:pkg-cli>\cli.toit:127:33
  3: main                      D:\a\toit\toit\tools\firmware.toit:120:12
Error: exit status 1
```

It works if downgrading to jaguar 1.7.3.

kasperl, 2022-11-25:
It looks like https://github.com/toitlang/toit/blob/feab0a3868a7fde5eed50017fa03891f09731c43/tools/firmware.toit#L496 is the problem.

kasperl, 2022-11-25:
@JWood48 We released v1.7.9 today as an attempt to work around the issue. Give it a spin!

kasperl, 2022-11-27:
Fixed in v1.7.9 🥇

closed 2022-11-27 by kasperl