toitlang/jaguar #312: Jag flash not working on Windows

robvanlopik, 2022-12-10:
The following says it all. (Windows 10)

Flashing device over serial on port 'COM4' ...
esptool.py v3.3.2-dev
Serial port COM4
Connecting...
Traceback (most recent call last):
  File "esptool.py", line 5399, in <module>
  File "esptool.py", line 5392, in _main
  File "esptool.py", line 4699, in main
  File "esptool.py", line 115, in get_default_connected_device
  File "esptool.py", line 664, in connect
  File "esptool.py", line 629, in _connect_attempt
  File "esptool.py", line 519, in sync
  File "esptool.py", line 458, in command
  File "esptool.py", line 421, in write
  File "serial\serialwin32.py", line 325, in write
serial.serialutil.SerialTimeoutException: Write timeout
[3444] Failed to execute script 'esptool' due to unhandled exception!
Error: exit status 1
PS C:\Users\robva\OneDrive\Code\Toit> jag version  
Version:         v1.7.12
SDK version:     v2.0.0-alpha.44
Build date:      2022-12-02T14:22:55Z

floitsch, 2022-12-12:
This is typically the consequence of one of:
- a bad USB cable
- bad serial port driver
- too high of a baud rate (although that's usually linked to other reasons, like running in a virtual machine...)

Do you know whether your hardware setup "should" be working? That is, have you maybe used your machine with another environment (like esp-idf, Arduino...) before?

robvanlopik, 2022-12-14:
After getting Thonny with micropython working with my ESP32's I tried again and now flashing works. Scan does not but scan ip-address finds the device and I can run a small program. I will close this issue and. In order to keep the issue list clean, I will address further questions to Discord

closed 2022-12-14 by robvanlopik

floitsch, 2022-12-14:
Do you maybe have an idea of what changed that could be the reason?
Was there anything non-Thonny related (like installing a fresh driver) that could have fixed the issue?

robvanlopik, 2022-12-14:
At some point the following port appeared: Silicon Labs CP210x USB to UART
Bridge.
This can also have occurred after connecting a Microbit device..