toitlang/jaguar #438: MacOS binary problems.

valientegaston, 2023-11-24:
I downloaded the latest version of jaguar for Macos (Monterrey) and I found some strange behaviors. To start the binary, when executing jag version, the execution does not end and it hangs waiting for something. In addition, it generates multiple stability problems in the operating system and I end up restarting it.
I have downloaded version 1.0.8 to test and I don't have that problem. The jag version command displays the information and finishes its execution appropriately. Although yes, when running a jag scan, it triggers an error since I don't have any device plugged in and after the error it hangs there.

This behavior is very strange and it is the first time it has happened to me in the years that I have been using MacOS.
I hope someone can reproduce this and we can find a solution.
I have also tried installing it using brew, but it reaches the make stage and hangs there.

Thank you very much for this tool, I am eager to try it.

floitsch, 2023-11-24:
Very strange indeed.

The Jaguar executable is a golang binary. In theory it should "just work".
I have looked at the brew step, and the `make` step really just builds the executable using `go`:

https://github.com/toitlang/homebrew-toit/blob/main/Formula/jaguar.rb#L24
https://github.com/toitlang/jaguar/blob/main/Makefile#L49

Do you know if golang works on your machine?

We haven't done big changes in Jaguar, so v1.0.8 shouldn't behave much different.

Could you show me the error you got after `jag scan`? Scanning is done using network (listening for UDP packets), and it shouldn't have an error if there aren't any devices. It should just come up empty handed.

valientegaston, 2023-11-24:
>Do you know if golang works on your machine?  Yes.  go v1.21.4 without any problem. 

**Please check this video...**    [https://youtu.be/R7kVqtyqJMs](https://youtu.be/R7kVqtyqJMs)

There is the comparison of the two versions window by window.
The older version works fine. The latest version crashes all the time, even just trying the basic "version" command.

The first run of version 1.21 does not even show the "version" content. On the second run it shows it but it hangs. Version 1.0.8 runs without problems many times.

Thank you so much!

valientegaston, 2023-11-24:
I forgot to comment, both video binaries were downloaded from the indicated page in this repository. I haven't compiled them myself. Since I can't do that either :)

erikcorry, 2023-11-24:
What does `uname -a` say about your Mac?

floitsch, 2023-11-24:
Could you also maybe try to build `jaguar` by hand:
```
mkdir debug-jaguar
cd debug-jaguar
git clone https://github.com/toitlang/jaguar
cd jaguar
mkdir build
go build -o build/jag cmd/jag/main.go
# See if it works.
build/jag version
```

In theory the brew script does pretty much the same thing, but it has a few more Go environment variables set.

valientegaston, 2023-11-24:
> What does `uname -a` say about your Mac?
gaston@iMac ~ % uname -a
Darwin iMac.local 21.6.0 Darwin Kernel Version 21.6.0: Thu Jul  6 22:18:26 PDT 2023; root:xnu-8020.240.18.702.13~1/RELEASE_X86_64 x86_64

valientegaston, 2023-11-24:
> Could you also maybe try to build `jaguar` by hand:
>
> ```
> mkdir debug-jaguar
> cd debug-jaguar
> git clone https://github.com/toitlang/jaguar
> cd jaguar
> mkdir build
> go build -o build/jag cmd/jag/main.go
> # [...]

gaston@iMac Toit % mkdir debug-jaguar
gaston@iMac Toit % cd debug-jaguar 
gaston@iMac debug-jaguar % git clone https://github.com/toitlang/jaguar
Cloning into 'jaguar'...
remote: Enumerating objects: 3140, done.
remote: Counting objects: 100% (521/521), done.
remote: Compressing objects: 100% (185/185), done.
remote: Total 3140 (delta 273), reused 490 (delta 251), pack-reused 2619
Receiving objects: 100% (3140/3140), 851.80 KiB | 1.57 MiB/s, done.
Resolving deltas: 100% (1679/1679), done.
gaston@iMac debug-jaguar % cd jaguar 
gaston@iMac jaguar % ls
CODE_OF_CONDUCT.md	Makefile		docs			package.lock		tools
CONTRIBUTING.md		README.md		go.mod			package.yaml
LICENSE			cmd			go.sum			src
gaston@iMac jaguar % mkdir build
gaston@iMac jaguar % go build -o build/jag cmd/jag/main.go
gaston@iMac jaguar % cd build
gaston@iMac build % ls
jag
gaston@iMac build % ./jag version
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x18 pc=0x16c604a]

goroutine 1 [running]:
github.com/toitlang/jaguar/cmd/jag/commands.JagCmd.VersionCmd.func3(0xc0001c8000?, {0x1ef8e40?, 0x4?, 0x184a5eb?})
	/Users/gaston/Toit/debug-jaguar/jaguar/cmd/jag/commands/version.go:30 +0x4a
github.com/spf13/cobra.(*Command).execute(0xc0001c8000, {0x1ef8e40, 0x0, 0x0})
	/Users/gaston/go/pkg/mod/github.com/spf13/cobra@v1.6.1/command.go:920 +0x863
github.com/spf13/cobra.(*Command).ExecuteC(0xc0001b4300)
	/Users/gaston/go/pkg/mod/github.com/spf13/cobra@v1.6.1/command.go:1044 +0x3a5
github.com/spf13/cobra.(*Command).Execute(...)
	/Users/gaston/go/pkg/mod/github.com/spf13/cobra@v1.6.1/command.go:968
github.com/spf13/cobra.(*Command).ExecuteContext(0x184bfb9?, {0x1970488?, 0xc00017d920?})
	/Users/gaston/go/pkg/mod/github.com/spf13/cobra@v1.6.1/command.go:961 +0x47
main.main()
	/Users/gaston/Toit/debug-jaguar/jaguar/cmd/jag/main.go:34 +0x190
gaston@iMac build %

valientegaston, 2023-11-24:
Result of Jag Scan with version 1.0.8

[image]

floitsch, 2023-11-24:
Ok. That looks like a "normal" error. I forgot we printed it instead of just saying that none were found.
It should, obviously, not hang afterwards...

floitsch, 2023-11-24:
Could you share the version of Go that is installed?

valientegaston, 2023-11-24:
> Could you share the version of Go that is installed?

gaston@iMac ~ % go version
go version go1.21.4 darwin/amd64

floitsch, 2023-11-24:
Nice. That's even more recent than mine...

I'm now trying to upgrade all Go packages. Specifically, it's probably a good idea to run the latest `x/sys`.
Will give you a branch to checkout once I have uploaded it. (Currently fighting `go get -u all`. It claims not to find a module).

floitsch, 2023-11-24:
Could you try again with the `floitsch/upgrade-go-deps` branch?

```
mkdir debug-jaguar-deps
cd debug-jaguar
git clone https://github.com/toitlang/jaguar
cd jaguar
git checkout floitsch/upgrade-go-deps  # <= new line.
mkdir build
go build -o build/jag cmd/jag/main.go
# See if it works.
build/jag version
```

valientegaston, 2023-11-24:
> build/jag version

gaston@iMac jaguar % git checkout floitsch/upgrade-go-deps                
branch 'floitsch/upgrade-go-deps' set up to track 'origin/floitsch/upgrade-go-deps'.
Switched to a new branch 'floitsch/upgrade-go-deps'
gaston@iMac jaguar % mkdir build
gaston@iMac jaguar % go build -o build/jag cmd/jag/main.go
gaston@iMac jaguar % build/jag version
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x18 pc=0x16eb66a]

goroutine 1 [running]:
github.com/toitlang/jaguar/cmd/jag/commands.JagCmd.VersionCmd.func3(0xc0002a0000?, {0x1f47ac0?, 0x4?, 0x1878094?})
	/Users/gaston/Toit/debug-jaguar/jaguar/cmd/jag/commands/version.go:30 +0x4a
github.com/spf13/cobra.(*Command).execute(0xc0002a0000, {0x1f47ac0, 0x0, 0x0})
	/Users/gaston/go/pkg/mod/github.com/spf13/cobra@v1.8.0/command.go:987 +0xaa3
github.com/spf13/cobra.(*Command).ExecuteC(0xc00028e300)
	/Users/gaston/go/pkg/mod/github.com/spf13/cobra@v1.8.0/command.go:1115 +0x3ff
github.com/spf13/cobra.(*Command).Execute(...)
	/Users/gaston/go/pkg/mod/github.com/spf13/cobra@v1.8.0/command.go:1039
github.com/spf13/cobra.(*Command).ExecuteContext(0x1879aa7?, {0x19a3758?, 0xc000255b90?})
	/Users/gaston/go/pkg/mod/github.com/spf13/cobra@v1.8.0/command.go:1032 +0x47
main.main()
	/Users/gaston/Toit/debug-jaguar/jaguar/cmd/jag/main.go:34 +0x190
gaston@iMac jaguar %

floitsch, 2023-11-24:
This might actually be a "valid" segfault caused by using a non-deployment build.
From looking at the sources, my guess is, that the `$HOME` variable is not set.

Could you try verify that `echo $HOME` is set (and ideally points to a valid folder)?

If it does, please try the `floitsch/upgrade-go-deps.fix-nil-sdk` branch. The segfault should go away, and it should print an error message.

valientegaston, 2023-11-24:
[image]

I am not an expert, but it see like GetSDK returns null and then you want sdk.Version. 
Thank you for your help. I really appreciate it

floitsch, 2023-11-24:
That's exactly my conclusion. As you can see, this branch is only triggered when `isReleaseBuild` is false, so you wouldn't usually enter this branch in released executables, but it's still iffy that `GetSDK` returns `nil`.

valientegaston, 2023-11-24:
[image]

This returns GetSDK, so I have to do a jag setup before I can do a jag version.

floitsch, 2023-11-24:
So maybe it actually works now (modulo the bug in the non-`isReleaseBuild` version.
Could you do a `jag setup` first and then `jag version` to see if things seem to behave nicely?

If yes I will mark the go-package update PR for review and we will release a new version soon.

valientegaston, 2023-11-24:
gaston@iMac build % ./jag version
Error: the path '/Users/gaston/.cache/jaguar/sdk' did not hold the SDK.
You must setup the SDK using 'jag setup'
gaston@iMac build % ./jag setup
Downloading Toit SDK from https://github.com/toitlang/toit/releases/download/v2.0.0-alpha.121/toit-macos.tar.gz ...
50.83 MiB / 50.83 MiB [------------------------------------------------------------------------] 100.00% 4.74 MiB p/s
Successfully installed Toit SDK into /Users/gaston/.cache/jaguar/sdk
Downloading Jaguar assets from https://github.com/toitlang/jaguar/releases/download/v1.21.0/assets.tar.gz ...
Successfully installed Jaguar assets into /Users/gaston/.cache/jaguar/assets
126.91 KiB / 126.91 KiB [-----------------------------------------------------------------------------] 100.00% ? p/s
Successfully setup Jaguar v1.21.0 with Toit SDK v2.0.0-alpha.121.
gaston@iMac build % ./jag version
Version:	 ---
SDK version:	 v2.0.0-alpha.121
Build date:	 unknown
gaston@iMac build %

It's working now :) Thank you!

valientegaston, 2023-11-24:
jag scan is working too.  No more internet lost. No more hangs :)

floitsch, 2023-11-24:
Good news!

It looks like the go-dependency updates made it work.
In that case, the next released version should work on your machine.

I'm pretty sure that the one you built also behaves nicely, so you should be able to use it until then.
If not, just drop a note here, or visit our discord.

Thanks so much for debugging this with me.

kasperl, 2023-11-25:
@valientegaston We just shipped Jaguar v1.22.0 with the fixes. Do let me know if this fixes the issues for you!

valientegaston, 2023-11-25:
@kasperl Unfortunately version 1.22 reproduces the same errors as 1.21
I'm going to try compiling 1.22 myself to see if it works correctly that way.
The latest version that I compiled by myself, using the floitsch/upgrade-go-deps.fix-nil-sdk branch, works without problems. I don't know why the pre-compiled version 1.22 doesn't do it.

floitsch, 2023-11-25:
We are building the executable with the following flags on the server:
`GODEBUG=netdns=go` (environment variable) and `-tags 'netgo osusergo'`.

I don't really know what these do, but otherwise it should be built the same...

valientegaston, 2023-11-25:
I have compiled from the repository, and the generated executable works without problems.
From what I can see, the executable I generate weighs 22.7MB, while the pre-compiled 1.22 weighs 22.9MB.
Now I try the options you mention above and tell you what happens.

valientegaston, 2023-11-25:
I think the problem happens when compiling with the -tags 'netgo osusergo' tag. I really don't know what that does, but if I compile with that tag, the strange errors that are in this post occur. If I compile without it, everything works fine. Maybe it's something about my Mac. When I have the opportunity to try it on another Mac, I'll upload the result here. Thank you very much for all the help.

floitsch, 2023-11-25:
I will investigate what that flag does. It's very likely we don't need it anymore.

floitsch, 2023-11-25:
I think it's safe to remove the custom Go flags, and I have uploaded a PR.

In the meantime: could you tell me whether the LSP server (IDE completion...) works for you? That's another Go executable that we ship with the Toit SDK. If it doesn't work for you, we might need to change the flags there too.

floitsch, 2023-11-25:
I have uploaded the output of the buildbot here: https://drive.google.com/file/d/16jGqQCZJYsB7tDZLAsEEUidsluok4nFg/view?usp=sharing

It's not yet signed, so you might need to call `codesign -fs - jag`.
Would be good to know if that version works now.

valientegaston, 2023-11-26:
Good news!! The version uploaded to the Drive link works without problems.

valientegaston, 2023-11-26:
toit.lsp seems to work. At least it doesn't generate any of the problems reported before, using jag

kasperl, 2023-11-26:
Preparing Jaguar v1.22.1 for release.

kasperl, 2023-11-26:
@valientegaston Jaguar v1.22.1 is out with the potential fix. I've tried it through Homebrew on my Mac and it works for me, but then again I never had any issues :)

Give it a spin and let us know if it works for you. Thanks!

valientegaston, 2023-11-27:
@kasperl Jaguar 1.22.1 works excellent. No problem so far. Thank you!

kasperl, 2023-11-28:
@valientegaston Thank you for confirming!

PR #440 (2023-11-25): Update go dependencies.
Fixes #438.
commit c9a344f9 (2023-11-25): Update go dependencies. (#440)

Fixes #438.
PR #443 (2023-11-26): Don't use custom Go flags.
Fixes #438.
commit 1729765e (2023-11-26): Don't use custom Go flags. (#443)

Fixes #438.