toitlang/jaguar #558: Jaguar unable to flash ESP32-S3 chip

yudistrange, 2024-09-23:
I have an ESP32-S3 chipset. I am trying to install jaguar on the chipset but failing with the following error. 

```
❯ jag flash -p /dev/ttyACM1 -c ESP32-S3

Downloading ESP32-S3 firmware from https://github.com/toitlang/envelopes/releases/download/v2.0.0-alpha.160/firmware-ESP32-S3.envelope.gz ...
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0xb2e6a8]

goroutine 1 [running]:
github.com/toitlang/jaguar/cmd/jag/commands.downloadGzipped({0xe089d8?, 0xc000199c50?}, {0xc000042a10?, 0xccf6a2?}, {0xc00007f3e0, 0x52})
	/home/runner/work/jaguar/jaguar/cmd/jag/commands/envelope.go:44 +0x48
github.com/toitlang/jaguar/cmd/jag/commands.downloadPublishedFirmware({0xe089d8, 0xc000199c50}, {0xc000040cc0, 0x10}, {0x7ffe28ed88b5, 0x8})
	/home/runner/work/jaguar/jaguar/cmd/jag/commands/envelope.go:136 +0x2ad
github.com/toitlang/jaguar/cmd/jag/commands.GetCachedFirmwareEnvelopePath({0xe089d8, 0xc000199c50}, {0xc000040cc0, 0x10}, {0x7ffe28ed88b5, 0x8})
	/home/runner/work/jaguar/jaguar/cmd/jag/commands/envelope.go:27 +0x105
github.com/toitlang/jaguar/cmd/jag/commands.FlashCmd.func1(0xc0001ecf08, {0xc0001ce840, 0x0, 0xccf6d2?})
	/home/runner/work/jaguar/jaguar/cmd/jag/commands/flash.go:97 +0x6e5
github.com/spf13/cobra.(*Command).execute(0xc0001ecf08, {0xc0001ce800, 0x4, 0x4})
	/home/runner/go/pkg/mod/github.com/spf13/cobra@v1.8.0/command.go:983 +0xaca
github.com/spf13/cobra.(*Command).ExecuteC(0xc0001e2608)
	/home/runner/go/pkg/mod/github.com/spf13/cobra@v1.8.0/command.go:1115 +0x3ff
github.com/spf13/cobra.(*Command).Execute(...)
	/home/runner/go/pkg/mod/github.com/spf13/cobra@v1.8.0/command.go:1039
github.com/spf13/cobra.(*Command).ExecuteContext(0xcd1106?, {0xe089d8?, 0xc000199c50?})
	/home/runner/go/pkg/mod/github.com/spf13/cobra@v1.8.0/command.go:1032 +0x47
main.main()
	/home/runner/work/jaguar/jaguar/cmd/jag/main.go:34 +0x18b
```

floitsch, 2024-09-23:
Try with `jag flash -p /dev/ttyACM1 -c esp32s3`

I will try to harden the Jaguar executable.

yudistrange, 2024-09-23:
Ah! Because I was using the capitalized name, it was going to the wrong link. Thanks!

floitsch, 2024-09-23:
Reopening: Jaguar shouldn't just crash when there is a bad chip.

yudistrange, 2024-09-23:
Fair.
Also I had copied the chip name from the error message that I got when I tried flashing without providing a chipset.

floitsch, 2024-09-23:
Confirmed. The error message is from the esptool.

yudistrange, 2024-09-23:
Hmm.. I guess we can have a supported list of chips [here](https://github.com/toitlang/jaguar/blob/7c423562566ac2b54c6bad58645a8ac8cf59a155/cmd/jag/commands/flash.go#L51)

And fail gracefully when the supplied chipset name doesn't match. What do you think?

floitsch, 2024-09-23:
I think I will special case the most common chip names and mistakes and otherwise invite the user to check that the chip name is correct.
Something like:
```
		if err := downloadPublishedFirmware(ctx, version, model); err != nil {
			switch model {
			case "esp32", "esp32s3", "esp32c3", "esp32s2":
				fmt.Printf("Failed to download firmware: %v\n", err)
			case "ESP32", "ESP32S3", "ESP32C3", "ESP32S2":
				fmt.Printf("Chip model names must be lowercase. Please try again with 'esp32', 'esp32s3', 'esp32c3', or 'esp32s2'.\n")
			case "ESP32-S2", "ESP32-S3", "ESP32-C3":
				fmt.Printf("Chip model names must be lowercase without dashes. Please try again with 'esp32s2', 'esp32s3', or 'esp32c3'.\n")
			default:
				fmt.Printf("Failed to download firmware: %v\n", err)
				fmt.Printf("Make sure the model name is correct. You can find supported models at\n")
				fmt.Printf("https://github.com/toitlang/envelopes/releases/tag/%s.\n", version)
			}
```

yudistrange, 2024-09-23:
Why not lower case the supplied chipset name before checking against the supported list anyway?
The names with `-` can fail as you suggested.

This is unrelated, but I see other envelopes as well such as `esp32s3-no-ble`.
When are they used? Can a user supply those as the chipset name? Or are they for internal use only?

floitsch, 2024-09-23:
> Why not lower case the supplied chipset name before checking against the supported list anyway? The names with `-` can fail as you suggested.
Agreed. I'm considering that.

> This is unrelated, but I see other envelopes as well such as `esp32s3-no-ble`. When are they used? Can a user supply those as the chipset name? Or are they for internal use only?

They can be used, and sometimes should be used. For example, if the ESP32 has SPIRAM, or octo-SPIRAM. See the README of the repository.

Since BLE uses memory even when not in use, it's beneficial to use that envelope if possible. Simply use `--chip esp32s3-no-ble`.

PR #559 (2024-09-24): Fix crash and provide better error when chip is wrong.
Fixes #558
commit 905c786a (2024-09-24): Fix crash and provide better error when chip is wrong. (#559)

Fixes #558
closed 2024-09-24 by floitsch