toitlang/jaguar #560: `jag run` doesn't work when device name or ID is supplied

yudistrange, 2024-09-23:
Jag run only succeeds when I supply the IP address of the device. It's not able to find the device based on name or device Id. My computer and the esp32 device are on the same network. I can ping the esp32 device. I can also see the jaguar device home page at 10.0.0.135:9000

### Relevant Logs
**Starting the monitor** 
```
❯ jag monitor -p /dev/ttyACM1
Starting serial monitor of port '/dev/ttyACM1' ...
Hello, Crap!
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403846a2
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x88
load:0x403c9700,len:0x96c
load:0x403cc700,len:0x2504
entry 0x403c9850
E (299) quad_psram: PSRAM ID read error: 0x00ffffff, PSRAM chip not found or not supported, or wrong PSRAM line mode
E (299) esp_psram: PSRAM enabled but initialization failed. Bailing out.
[toit] INFO: starting <v2.0.0-alpha.160>
[toit] DEBUG: clearing RTC memory: powered on by hardware source
[toit] INFO: running on ESP32S3 - revision 0.2
[wifi] DEBUG: connecting

[wifi] DEBUG: connected
[wifi] INFO: network address dynamically assigned through dhcp {ip: 10.0.0.135}
[wifi] INFO: dns server address dynamically assigned through dhcp {ip: [64.71.255.204, 64.71.255.198]}
[jaguar.http] INFO: running Jaguar device 'sharp-bank' (id: '634f6df6-ef53-46f8-811d-5672179f75d3') on 'http://10.0.0.135:9000'
```

**Jag Run with Device Name**
```
❯ jag run -d 'sharp-bank'  hello.toit
Scanning for  device with name: 'sharp-bank'
Error: didn't find any Jaguar devices.
Perhaps you need to be on the same wifi as the device.
You can also specify the IP address of the device
```

**Jag Run with Device ID**
```
❯ jag run -d '634f6df6-ef53-46f8-811d-5672179f75d3'  hello.toit
Scanning for  device with ID: '634f6df6-ef53-46f8-811d-5672179f75d3'
Error: didn't find any Jaguar devices.
Perhaps you need to be on the same wifi as the device.
You can also specify the IP address of the device
```

### Successful Run when using the IP address
```
❯ jag run -d '10.0.0.135'  hello.toit
Scanning for  device with address: '10.0.0.135'
Running 'hello.toit' on 'sharp-bank' ...
Success: Sent 33KB code to 'sharp-bank'
```

kasperl, 2024-09-24:
Thanks for the report, @yudistrange. Is it possible that you have some software firewall that blocks UDP broadcast messages?

yudistrange, 2024-09-24:
Oh that's quite possible. I'll check and fix that.
Thanks!

closed 2024-09-24 by yudistrange