toitlang/jaguar #561: Support getting wifi configuration

addshore, 2024-09-27:
We have a device which has had a wifi config set, we would like to retrieve the config to see what has been set (primmarily to compare the settings to another ESP device).

a `jag config wifi get` command would be appreciated if possible.

kasperl, 2024-09-27:
How would you like to retrieve the config? It is possible to access the wifi configuration from code on the device, but it sounds like maybe you need `jag config wifi get` to talk to the device without necessarily being on the same network.

floitsch, 2024-09-27:
The `jag config` commands only work locally. You could just look at the configuration file in `$HOME/.config/jaguar/config`.

If you have access to the device through Jaguar you could run a program that gets it from the device: 

```
import system.firmware

main:
  print firmware.config["wifi"]
```

addshore, 2024-09-27:
You are both correct indeed, 

This is probably answer enough for me!

> If you have access to the device through Jaguar you could run a program that gets it from the device:
> 
> ```
> import system.firmware
> 
> main:
>   print firmware.config["wifi"]
> ```

closed 2024-09-27 by addshore