toitlang/jaguar #585: Issue with empty prefixes and packages being provided in package.lock file

addshore, 2025-01-13:
I have the following file, whcih I believe was auto generated at some point.

```
sdk: ^2.0.0-alpha.145
prefixes:
packages:
```

While trying to `jag run` a simple script, jag complains aboute this file as it expects a map

```
jag run ./pkg/pupa/src/V3Data.test.toit
Running './pkg/pupa/src/V3Data.test.toit' on 'previous-present' ...
......../package.lock:2:10: error: Invalid package lock file. Expected a map, got a scalar
prefixes:
         ^
Compilation failed
```

I imagine the reading code just needs to be adjusted to allow this empty case?

floitsch, 2025-01-14:
Hmm.
Not sure how that file could have been created.
'prefixes' and 'packages' can indeed be empty, but they still need to be maps. I think for yaml that would mean:
```
prefixes: {}
packages: {}
```
We could add a special case for when the entry is the empty string, but that feels weird.

floitsch, 2025-01-14:
From what I can see, our tools remove empty `prefixes` and `packages` sections.

closed 2025-01-14 by addshore